BlokasLabs/USBMIDI #2: output

comarius, 2018-04-22:
how do I listen to it. plugging in to a midi input, aka Yamaha, does not pop up as a midi device?

gtrainavicius, 2018-04-22:
Hi, what are the models of devices you are talking about and which OS?

comarius, 2018-04-23:
HI,
I wanted to plug it to yamaha-61 in the USB port. I thought that USB would be presented as a MIDI interface, but is still seen as /dev/ttyUSB serial.

gtrainavicius, 2018-04-23:
I find multiple models of Yamaha keyboard with 61 keys, what is the model number?

Also, what Arduino board are you using?

gtrainavicius, 2018-05-04:
@comarius ping :)

gtrainavicius, 2018-05-13:
Closing this issue due to inactivity. If the OS used was Linux based, then `amidi` and `aconnect` (or `aconnectgui`) should allow sending and receiving MIDI data between OS and Arduino.

closed 2018-05-13 by gtrainavicius