BlokasLabs/USBMIDI #3: Example code fails to build

oofus, 2018-10-31:
Hi. Trying to build the example code for USBMIDI and it's failing with the following message

src\usbmidi_pluggableusb.cpp:90:2: error: 'Fifo' does not name a type

  Fifo m_midiInFifo;

Seems not to be using the template to define the type, and fifo.h isn't included in usbmidi_pluggableusb.cpp.

This is V1.1.0 installed from within the Arduino IDE.

gtrainavicius, 2018-10-31:
Hey, what board are you targeting?

gtrainavicius, 2018-10-31:
Ok, I have just fixed the issue, tagged a v1.1.1 release. It should get picked up automatically by Arduino IDE eventually, but to get the fix sooner, just remove v1.1.0 USBMIDI library and do:

```bash
cd .../Arduino/libraries
git clone https://github.com/BlokasLabs/USBMIDI/
```

I've tested this Pluggable USB implementation with SparkFun ProMicro, works fine now. :)

Thank you for reporting the issue!

oofus, 2018-10-31:
That has fixed the build issue, thanks. Now to test and write some of my own code !

closed 2018-10-31 by gtrainavicius