BlokasLabs/USBMIDI #4: USBMIDI does not compile on Arduino Due ...

kaspencer, 2018-11-06:
I need to read some Sysex and Sysex Raw messages into an Arduino Due. I have worked with Sysex messages in the Arduino Teensy with no problems, but my current project requires many pins of which the Due has sufficient.
My efforts with the Due are generating two issues:
1. When I try to use the USBMIDI library on the Arduino Due it fails owing to the PluggableUSB.h file being not found. What is the solution to this?
2. Will the USBMIDI library retrieve a complete Sysex message using a usbMIDI.read() function (or similar) as per the Teensy usbMIDI library?
Many thanks - especially for the Library devlopment and hopefully for an answer to the issues.
KA Spencer

gtrainavicius, 2018-11-06:
Hi, turns out the ARM cores have a very similar but slightly incompatible PluggableUSB.h unfortunately, and it's additionally in "USB" subfolder within the Arduino core... I have created this issue for the Arduino project regarding this inconsistency: https://github.com/arduino/ArduinoCore-sam/issues/62

I was able to modify the USBMIDI library so it compiles for Arduino Due, however, I don't own any such ARM based Arduino board, so I can't say that my changes are the correct ones. You can find them in `sam` branch of this repository:

https://github.com/BlokasLabs/USBMIDI/tree/sam

You may try removing USBMIDI folder from your arduino/libraries, and checking out the `sam` branch into arduino/libraries/USBMIDI

The only place that I am not sure about is this one: https://github.com/BlokasLabs/USBMIDI/commit/d2985d078225b8da556edc633a5bce562aa79572#diff-78597e189949cd6f2eb860b1b1db86f3R100 - it is different from AVR PluggableUSB.h in an unobvious way...

gtrainavicius, 2018-11-06:
Regarding sysex - this library provides and consumes data in exactly the same API as you'd be sending MIDI via Serial. So in case of sysex, your code would have to detect 0xf0 (sysex start), buffer or handle the incoming data as it comes in, and detect 0xf7 (sysex end) message to know that the message has ended.

kaspencer, 2018-11-06:
Thanks gtgtrainavicius for the help with both points.
I have to pursue this so I will take up and try both of your suggestions. Maybe you won't mind if I post any comments after that, should I feel it worthwhile.
I must say that the Due seems tricky in several respects, especially compared to the Teensy (3.2) which I have had running for quite while now doing its bit with MIDI Sysex Raw messages. But for the current project I just need many more pins.
Using the Native Port (which you have to do for MIDI over USB) on the Due is fraught as SerialUSB is also currently refusing to work.
We'll have to keep on eh!
Best wishes.
Kenneth Spencer

gtrainavicius, 2018-11-06:
Sure, if you manage to get USBMIDI working on the board, let me know! I'll integrate the necessary changes for his board to `master` :)