BlokasLabs/USBMIDI #6: Attiny45: usbdrvasm165.inc:53: Error: constant value required

Harvie, 2019-01-07:
I've been trying to compile this using vusb for digispark (atmega45 with micronucleus bootloader).
It seems to me, that some definition is missing:

Line 53 of usbdrvasm165.inc:    `sbis    USBIN, USBMINUS`

It gives following errors:

```
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc: Assembler messages:
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:53: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:57: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:59: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:61: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:63: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:65: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:67: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:92: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:111: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:140: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:145: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:150: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:157: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:163: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:170: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:177: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:187: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:192: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:198: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:205: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:217: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:222: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:231: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:242: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:250: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:256: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:263: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:271: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:277: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:283: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:290: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:296: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:302: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:308: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:314: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:320: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:327: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:368: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:391: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:393: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:394: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:395: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:404: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:417: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:428: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:440: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:448: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:449: Error: constant value required
/home/harvie/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm165.inc:450: Error: constant value required
exit status 1
```

Harvie, 2019-01-07:
vusb config should look like this, but i am not sure where to put it and if that's enough

```
USB_CFG_IOPORTNAME=B
USB_CFG_DMINUS_BIT=3
USB_CFG_DPLUS_BIT=4
```

gtrainavicius, 2019-01-08:
For setting up your config for pins, either edit usbconfig.h locally, or set up a custom board definition for Arduino, for example, see here: https://github.com/BlokasLabs/Blokas-Boards/blob/master/avr/boards.txt, at the bottom of the file there's -D... arguments to set the preprocessor defines. Boards.txt would be the best option if you want to support multiple boards.

I can't figure out why USBMIDI wouldn't work with Digispark, could you post your sketch and any modifications you've already made, also which board you have selected in Arduino IDE?

Harvie, 2019-01-08:
Sketch does not really matter, since USB device does not even enumerate (once it leaves micronucleus bootloader, which enumerates OK), even simple stuff like this:

```C
#include <usbmidi.h>

void setup() {
}

void loop() {
  while(USBMIDI.available()) USBMIDI.read();
}
```

Basicaly only change i've done was setting this in usbconfig.h unconditionaly:

```
#define USB_CFG_IOPORTNAME B
#define USB_CFG_DMINUS_BIT 3
#define USB_CFG_DPLUS_BIT 4
```

In arduino IDE use `Digispark (default - 16.5MHz)` board from following repository:
`http://digistump.com/package_digistump_index.json`

Using this setup i am able to get following USB MIDI library to work:
https://github.com/heartscrytech/DigisparkMIDI

But it can only send MIDI from uC to computer, so i would preffer to use USBMIDI of yours.

gtrainavicius, 2019-01-08:
I suspect the USB init code difference between what [DigisparkMIDI does](https://github.com/heartscrytech/DigisparkMIDI/blob/master/DigiMIDI.h#L264) and what [this library does ](https://github.com/BlokasLabs/USBMIDI/blob/master/src/usbmidi_vusb.cpp#L259)could be causing the issue.

Maybe `midiUsbInit` should make usbDeviceDisconnect call regardless of whether `USB_CFG_PULLUP_IOPORT` is defined, and have a much bigger delay (like 500ms) before reconnecting USB and calling usbInit.

I don't have an ATTinyXX chips available to try and set up a board similar to Digispark one at the moment, but if you could, it'd be great if you'd experiment with `midiUsbInit`, trying to make it more similar to what the other MIDI lib does and seeing if it fixes the issue.

Harvie, 2019-01-08:
i even added preprocessor debug to sketch:

```
/* definition to expand macro then apply to pragma message */
#define VALUE_TO_STRING(x) #x
#define VALUE(x) VALUE_TO_STRING(x)
#define VAR_NAME_VALUE(var) #var "="  VALUE(var)

/* Test values */
#pragma message(VAR_NAME_VALUE(USB_CFG_IOPORTNAME))
#pragma message(VAR_NAME_VALUE(USB_CFG_DMINUS_BIT))
#pragma message(VAR_NAME_VALUE(USB_CFG_DPLUS_BIT))
```
And that indicates the defines are set properly:

```
UsbMidiConverter.ino:26:51: note: #pragma message: USB_CFG_IOPORTNAME=B
 #pragma message(VAR_NAME_VALUE(USB_CFG_IOPORTNAME))
                                                   ^
UsbMidiConverter.ino:27:51: note: #pragma message: USB_CFG_DMINUS_BIT=3
 #pragma message(VAR_NAME_VALUE(USB_CFG_DMINUS_BIT))
                                                   ^
UsbMidiConverter.ino:28:50: note: #pragma message: USB_CFG_DPLUS_BIT=4
 #pragma message(VAR_NAME_VALUE(USB_CFG_DPLUS_BIT))
                                                  ^
```

Harvie, 2019-01-08:
> it'd be great if you'd experiment with midiUsbInit, trying to make it more similar to what the other MIDI lib does and seeing if it fixes the issue.

Ok i may take a look at it...

Harvie, 2019-01-08:
I've added following two lines, but it's still not enumerating :-/

```
static void midiUsbInit(void)
{
        // Activate pull-ups except on USB lines.
        USB_CFG_IOPORT = (uint8_t)~((1 << USB_CFG_DMINUS_BIT) | (1 << USB_CFG_DPLUS_BIT));

        usbDeviceDisconnect(); // HERE ////////////////////////////////////////////////////////////////////////////////
        // All pins input except USB (-> USB reset).
#ifdef USB_CFG_PULLUP_IOPORT // Use usbDeviceConnect()/usbDeviceDisconnect() if available.
        USBDDR = 0;              // We do RESET by deactivating pullup.
        usbDeviceDisconnect();
#else
        USBDDR = (1 << USB_CFG_DMINUS_BIT) | (1 << USB_CFG_DPLUS_BIT);
#endif

        // USB Reset by device only required on Watchdog Reset.
        uint8_t j = 0;
        while (--j) {   /* USB Reset by device only required on Watchdog Reset */
                uint8_t i = 0;
                while (--i);  /* delay >10ms for USB reset */
        }

        usbDeviceConnect(); // HERE ////////////////////////////////////////////////////////////////////////////////
#ifdef USB_CFG_PULLUP_IOPORT
        usbDeviceConnect();
#else
        USBDDR = 0; // Remove USB reset condition.
#endif

        usbInit();
}

```

gtrainavicius, 2019-01-08:
Try increasing the delay. There was some reason why _delay_ms was not used, but can't recall it, try placing ` _delay_ms(500);` before usbDeviceConnect. (may have to also `#include <avr/delay.h>` at the top)

Harvie, 2019-01-08:
I tried both. Adding one more nested loop and using _delay_ms(1000).
Didn't helped...

```
        // USB Reset by device only required on Watchdog Reset.
        uint8_t j = 0;
        while (--j) {   /* USB Reset by device only required on Watchdog Reset */
                uint8_t i = 0;
                while (--i) {  /* delay >10ms for USB reset */
                        uint8_t ij = 0;
                        while (--ij);  /* delay >10ms for USB reset */
                }
        }

        _delay_ms(1000);
```

Harvie, 2019-01-08:
I also tried wrapping the code with `cli();` and `sei();` like they do...

Harvie, 2019-01-08:
The exact same code does not work either...

```C
static void midiUsbInit(void)
{
        cli();
        usbDeviceDisconnect();
        _delay_ms(500); //250
        usbDeviceConnect();

        //wdt_enable(WDTO_1S);
        usbInit();

        sei();
}

```

gtrainavicius, 2019-01-08:
Are you calling `USBMIDI.poll()` in your loop()?

Harvie, 2019-01-08:
No. At least not directly. Tried to add, but didn't helped:

```
#include <usbmidi.h>

void setup() {
}

void loop() {
  USBMIDI.poll();
  while(USBMIDI.available()) USBMIDI.write(USBMIDI.read());
  USBMIDI.flush();
}
```

Harvie, 2019-01-08:
But it does not work even with empty setup() and loop()

Harvie, 2019-01-08:
> Please try putting: at the bottom of your usbconfig.h

WOW! That actually did the trick!

Harvie, 2019-01-08:
What do you think about including something like this to your usbconfig.h? That way there will be quite good chance that it USBMIDI will properly compile without editing boards.txt

```C
#ifndef USB_CFG_DMINUS_BIT

#if defined (__AVR_ATtiny44__) || defined (__AVR_ATtiny84__)
#define USB_CFG_IOPORTNAME      B
#define USB_CFG_DMINUS_BIT      1
#define USB_CFG_DPLUS_BIT       2

#elif defined (__AVR_ATtiny45__) || defined (__AVR_ATtiny85__)
#define USB_CFG_IOPORTNAME      B
#define USB_CFG_DMINUS_BIT      3
#define USB_CFG_DPLUS_BIT       4

#elif defined (__AVR_ATtiny87__) || defined (__AVR_ATtiny167__)
#define USB_CFG_IOPORTNAME      B
#define USB_CFG_DMINUS_BIT      3
#define USB_CFG_DPLUS_BIT       6

#elif defined (__AVR_ATtiny461__) || defined (__AVR_ATtiny861__)
#define USB_CFG_IOPORTNAME      B
#define USB_CFG_DMINUS_BIT      5
#define USB_CFG_DPLUS_BIT       6
#else
/*	ATtiny2313, ATmega8/48/88/168	*/
#define USB_CFG_IOPORTNAME      D
#define USB_CFG_DMINUS_BIT      3
#define USB_CFG_DPLUS_BIT       2
#endif

#endif
```

Harvie, 2019-01-08:
For the record: i've just tested with fresh copy of USBMIDI 1.1.1 installed through arduino IDE.
All i needed to add in order to work on attiny85 digispark were these magic lines:

```C
#define USB_CFG_IOPORTNAME      B
#define USB_CFG_DMINUS_BIT      3
#define USB_CFG_DPLUS_BIT       4

#define USB_INTR_CFG            PCMSK
#define USB_INTR_CFG_SET        (1<<USB_CFG_DPLUS_BIT)
#define USB_INTR_ENABLE_BIT     PCIE
#define USB_INTR_PENDING_BIT    PCIF
#define USB_INTR_VECTOR         SIG_PIN_CHANGE
```

gtrainavicius, 2019-01-09:
> WOW! That actually did the trick! Thank you very much.

Great! Could you try reverting midiUsbInit() and any other changes to original, and see whether usbconfig.h changes were the only ones necessary to get it working?

> Now i wonder, how we can work this into the USBMIDI repository in way that it does not break compiling for other boards.

The problem with preprocessor conditionals based only on the MCU type is that the same type of MCU can appear on many different boards, with different pinouts used, so this is not good enough.

Looks like builds for Digispark board define ARDUINO_AVR_DIGISPARK, this, as well as MCU type defines like \_\_AVR_ATtiny45__, could be used to predefine the variables for usbconfig.h

However, such a config file would take a lot of time and resources to maintain which I can't do on my own. I would gladly accept a contribution though.

The board preconfiguration file would have to follow these priority rules:

1. If a variable is defined using -D... command line option, use that value
2. If exact board and MCU was identified, use the value meant for this combination
3. Use default value

A quick example on how it could look like for one variable:

usbboard.h:
```c
#ifdef ARDUINO_AVR_DIGISPARK

#   ifndef USB_CFG_DMINUS_BIT
#       if defined (__AVR_ATtiny44__) || defined (__AVR_ATtiny84__)
#           define USB_CFG_DMINUS_BIT 1
#       elif defined (__AVR_ATtiny45__) || defined (__AVR_ATtiny85__)
#           define USB_CFG_DMINUS_BIT 3
//#     elif ...
#       endif
#   endif // USB_CFG_DMINUS_BIT

// Same for the rest of the variables that must be defined for DIGISPARK boards.
// If the board and MCU combination is not recognized, the variable should be left undefined.

#endif // ARDUINO_AVR_DIGISPARK
```

Then usbconfig.h could `#include "usbboard.h"` somewhere near the top.

gtrainavicius, 2019-01-09:
> For the record: i've just tested with fresh copy of USBMIDI 1.1.1 installed through arduino IDE.
All i needed to add in order to work on attiny85 digispark were these magic lines:

Ok, good, was about to ask for this information. :)

Harvie, 2019-01-09:
But i don't really know, how to include that `USB_INTR_*` defines... I can't really tell how common that is... But if we figure out how to detect digispark i would like to enable at for digispark, as it's very frequent and affordable board.

Harvie, 2019-01-09:
> It should have at least USBMIDI.poll() in the loop to do the USB work.

I've just read that in README.md, usualy i learn from library examples and none of the examples has poll() call in it. That's why i didn't included it.

I guess this is only needed for V-USB, but not native USB (LUFA), is that true? I think we should mention this in readme...

gtrainavicius, 2019-01-10:
> I guess this is only needed for V-USB, but not native USB (LUFA), is that true? Also as long as you call flush() it has poll() call in it... I think we should mention this in readme...

For most portability, it should be called regardless of underlying implementation.

Its absence from examples is my mistake, will fix them soon. :)

> But i don't really know, how to include that USB_INTR_* defines... I can't really tell how common that is... But if we figure out how to detect digispark i would like to include it, as it's very frequent and affordable board.

See the code example I posted above - `#ifdef ARDUINO_AVR_DIGISPARK` is for detecting 'Digispark' board.

The usbconfig customizations for Digispark may be borrowed from DigisprakMIDI's usbconfig.h, there's variables at the [top](https://github.com/heartscrytech/DigisparkMIDI/blob/master/usbconfig.h#L29) and [bottom](https://github.com/heartscrytech/DigisparkMIDI/blob/master/usbconfig.h#L393).

I'd really appreciate if you could extend the above config example for the rest of the variables required for Digispark boards and give it a spin. The code may be formatted differently, the only important thing is to respect the priority rules defined above. :)

Harvie, 2019-01-10:
> See the code example I posted above

Should i prepare PR, or you can do it yourself? :-)

gtrainavicius, 2019-01-14:
A PR would be great, also you could try it out immediately.

Otherwise I can make the changes towards the end of this week and ask you to test it out.

Harvie, 2019-01-14:
> you could try it out immediately.

OK. makes sense.

Harvie, 2019-01-15:
I think we can close this, but there is just last note:

If you add mention of digispark support to library.properties and README.md, that might get you few more users...

gtrainavicius, 2019-01-15:
Hey, I just did that :) Also tagged a new version, so Arduino IDE's library manager will pick up the recent changes.

Thank you for your help!

closed 2019-01-15 by gtrainavicius