BlokasLabs/USBMIDI #8: USBMIDI doesn't compile on Arduino Mini Pro 3.3V 8MHz

stephaneAG, 2019-04-29:
Hi there !
Sorry for comnig back on a subject closely related to the closed issue on ATmega328P & V-USB ( https://github.com/BlokasLabs/USBMIDI/issues/1 )

I just tried compiling for Aruiduino UNO -> went fine, for Arduino Mini Pro 5V,16MHz -> went fine, but I get the following error when trying to do so for the Arduino Mini Pro 3.3V,8MHz:

```
In file included from /Users/philippegarnier/Documents/Arduino/libraries/USBMIDI/src/vusb.S:18:0:
/Users/philippegarnier/Documents/Arduino/libraries/USBMIDI/src/../vusb/usbdrv/usbdrvasm.S:390:9: error: #error "USB_CFG_CLOCK_KHZ is not one of the supported non-crc-rates!"
 #       error "USB_CFG_CLOCK_KHZ is not one of the supported non-crc-rates!"
```

Not seeing anything related to it within 'usbboard.h', I wonder if this is even supported by V-USB ?
-> I'll digg the subject while trying using a 5V one if it's sure to work, but just posting this here in case someone else also wonders or even has the answer ;)

big thanks for the lib: can't wait to see it working ( in combo with ADCTouch ;p )
+

gtrainavicius, 2019-04-29:
Hey, if you may pick the Arduino board you use, pick something that's based on ATtmega32U4, like SparkFun Pro Micro. It has a USB peripheral built in, so no additional circuit is necessary for USB.

Otherwise, you have to choose the correct GPIO pins and build a small circuit between the microcontroller and the USB. Check out the V-USB library website (https://www.obdev.at/products/vusb/index.html) for example projects, as well as the library contents: https://github.com/BlokasLabs/USBMIDI/tree/master/vusb/circuits

stephaneAG, 2019-04-30:
hello :)
I got few Arduino micros on their way, yet I just gathered the necessary components for the circuit exposed at: https://web.archive.org/web/20160317191623im_/http://practicalarduino.com/sitebuilder/projects/knowledge/asset/medium/7/virtual-usb-keyboard-schematic-small.jpg

I'm wondering which pin is DIO5 in the above schematic, but it also seems the pins used have changed ( I only see 2 pins in the later circuits: both yours & the ones on the v-usb website )

I currently plan on using the zener method with a 5V 16MHz mini pro*
( so either the circuit in the linked schematic or https://github.com/BlokasLabs/USBMIDI/blob/master/vusb/circuits/with-zener.png )

A little question, though: the v-sub website uses PD3 & USBMIDI PD4, any technical reason for that ?
-> this being said, I still have to check if I can set other pins to be used or if I have to mod my current code ( .. )

anyway, thanks for the quick answer ;)

*From what I could read, I could also change the voltage regulator on it & under-power it while still getting it to work ( although maybe in a possibly unreliable manner .. )

gtrainavicius, 2019-05-02:
The DIO5 is an optional "pull up" pin which can be used to indicate whether the microcontroller is attached and ready to work with the USB bus. It's useful for example when you do a software reset, you'd retrigger the pin, to indicate to the other side that the communication has restarted. If you don't need such functionality, it can be prewired, as in the 'with-zener.png'

> A little question, though: the v-sub website uses PD3 & USBMIDI PD4, any technical reason for that ?
-> this being said, I still have to check if I can set other pins to be used or if I have to mod my current code ( .. )

You can use any digital pins, as long as they have either INTn, or PCINTn kind of interrupts. usbconfig.h must be set up accordingly for the pin you use, you may have to configure this part as well: https://github.com/BlokasLabs/USBMIDI/blob/master/src/usbconfig.h#L388

> *From what I could read, I could also change the voltage regulator on it & under-power it while still getting it to work ( although maybe in a possibly unreliable manner .. )

Check the 'Speed Grades' in the datasheet of the microcontroller you plan to use. Here's the graph of safe operating speed vs supply voltage for ATmega328:

[image]

closed 2019-05-13 by gtrainavicius