BlokasLabs/USBMIDI #11: Aditional MCUs

oofus, 2020-03-29:
Hi
Wasn't sure how else to ask a question about this lib other than here!

What's the likelyhood of support for other MCUs beyond AVR, particularly STM32 ? (or anything else 32bit for that matter)

Thanks

gtrainavicius, 2020-03-29:
Which boards do you have in mind?

This library depends on the 'PluggableUSB' library in Arduino core - it should allow to have a consistent API between architectures, but unfortunately the library somehow ended up being inconsistent between the architectures itself. :)

There is a 'sam' branch available on this repository that has changes required to make it run on Atmel samd based boards, see this pull request for a bit more info: https://github.com/BlokasLabs/USBMIDI/pull/10

So if your STM32 based Arduino board has a proper PluggableUSB implementation available, USBMIDI should work.

oofus, 2020-03-30:
Thanks for the reply, that's food for thought. I've just started looking at the whole range of STM32 devices, they seem quite compelling considering their capability and cost. I'll Investigate PluggableUSB on those devices.

Interesting about the Atmel SAM branch.

I'll do some experimentation and report back anything of interest.

Thanks

oofus, 2020-03-31:
From what I can see, STM32 Arduino Core does not use PlugableUSB, so that's a non-starter.

gtrainavicius, 2020-03-31:
A feature request on the STM32 Arduino Core library may help the devs consider implementing Pluggable USB support. :)

oofus, 2020-04-11:
Seems they plan to, but there has been no progress so far. Which is slightly disappointing.