BlokasLabs/USBMIDI #13: Understanding poll()

oofus, 2020-06-25:
Hi

I might be missing something really obvious, but to aid my understanding, could you explain why it is necessary to call USBMIDI.poll() regularly, or at all. I noticed that one of the first things USBMIDI.available() does is call _poll(). Doesn't that make USBMIDI.poll() redundant ? Removing USBMIDI.poll() from my code doesn't stop it working.

Thanks

gtrainavicius, 2020-06-26:
It's more like a safeguard, it must end up getting called quite often, so the USB communication is not lost. Especially in case the main loop is built in a way that does not read the input often enough. Of course ideally you should read the input as soon as possible.

Depending on it getting called inside the available method could also break some time in the future, if in some future version it gets removed. So this should be considered as an API contract when using this library, that USBMIDI.poll should be called at least once in the loop.

oofus, 2020-06-26:
Ahh, ok, understood. My code has USBMIDI.poll() and USBMIDI.available() in the same loop so are called as frequently as each other, hence my train of thought. I see now that might not be the case in other designs.

Thanks for the clarification.

closed 2020-06-26 by gtrainavicius