BlokasLabs/USBMIDI #15: USBMIDI.write() causes delay

tapir, 2021-04-21:
I have tried it with the `midictrl` example. Pressing the button takes 1-2 seconds. I have isolated the problem to USBMIDI.write() function. Once I comment it out, the rest of the code runs instantly without any delay.

Now, I don't believe this is an issue with this library specifically because MIDIUSB gives exactly the same problem with a similar minimal sketch. I'm opening this issue in the hopes that you might have some idea what's going on. 

My board is a clone of Leonardo. I have designed it from scratch but apart from MIDI, serial and flashing works without a problem so I want to believe the problem is not with my design however it might very well be. 

Any pointer would be much appreciated
Thank you

gtrainavicius, 2021-04-21:
USBMIDI.write may block in case the write buffer is full, until the host it's connected to reads the data from the Arduino device. The output buffer is 64 bytes long. The first few clicks you shouldn't see any delays.

Make sure there's some software running on the host to consume the incoming data.

tapir, 2021-04-21:
Oh I'm very dumb to think that somehow the buffer would be discarded when full. Thanks this explains all.

closed 2021-04-21 by tapir