BlokasLabs/USBMIDI #16: Unable to compile for Digispark

phesterman, 2021-09-03:
I'm having trouble compiling the examples for a Digispark.  I've tried using both the ATTinyCore and Digistump cores. The compiler doesn't seem to recognize 'u8' as a datatype, but treats it as if it's a variable name.  I changed 'u8' to 'byte' to see what would happen...using the Digistump core the compiler then gives an error the memory limits were exceeded; using the ATTinyCore, there were too many compiler errors to list—obviously, that wasn't the solution!  I'm on a Mac running Big Sur and Arduino 1.8.15; I loaded the USBMIDI library using Library Manager.  Is there another library I'm missing?

Thanks.  Peter

elektrouwe, 2022-02-22:
Peter, same problems here. Also tried to compile it for an UNO - also no success. (WIN10, Arduino 1.8.13)
could you get it running meanwhile ? 
Uwe

phesterman, 2022-02-22:
Uwe,
No luck.  In the end, I just gave up.
Peter