BlokasLabs/USBMIDI #17: potentiometer floating, unstable

jovabra, 2021-11-15:
potentiometer floating, unstable, using code **midictrl.ino**
what can be wrong? Leonardo plaque. **Thanks**

gtrainavicius, 2021-11-15:
How unstable is it? Is it jumping between 2 close-by values?

jovabra, 2021-11-15:
gtrainavicius, Thanks, I was actually checking AnalogRead, readCC and others. Now a curiosity, does it really have to be 10k Potentiometer? Could be 20k, no?

gtrainavicius, 2021-11-15:
The potentiometer acts as a resistor voltage divider network. 20k would probably work ok. Depending on the speed the ADC is reading, higher resistance values may provide wrong results, as it takes some time to charge up the ADC pin during the sampling phase. Lower resistance values consume more power, but charge the pin quicker. Generally, it looks like it's recommended to have 10k impedance for ATmega inputs, see this discussion:

https://electronics.stackexchange.com/questions/107741/required-output-impedance-for-adc-input

jovabra, 2021-11-15:
I get it... I figured the 5v Voltage is the same that feeds the Arduino. JUST not.... hahaha. The 5V in the case of Leonardo, will guarantee the stability of the ADC (Digital Analog Converter) Through the sampling voltage that is filtered by RC.

Well, in this case, the Impedance really shouldn't go outside the 10k spec, or less. There is probably my problem. Thank you so much, gtrainavicius, forum sharing, it was great.

gtrainavicius, 2021-11-16:
Cheers! :)

closed 2021-11-16 by gtrainavicius