BlokasLabs/USBMIDI #18: Can't get input working with Uno R3 board

Batophobia, 2022-02-08:
I'm using the UsbMidiSynth example, just replacing the tone/noTone calls (lines 68/69) with turning on/off the onboard LED, i.e. digitalWrite(LED_BUILTIN, HIGH);

I've got the Uno connected directly to the USB-B jack on a KORG B2.  A test blink in startup works fine, but playing any notes doesn't seem to turn the light on.  Even adding a line to turn it on at the top of the while (USBMIDI.available()) loop doesn't seem to work.

My first thought was that the baudrate was off, but I'm not sure how to determine what it should be outside of random guessing, and I'm not gonna go through hundreds of thousands of options.  So far I've tried some values from various MIDI examples [115200, 31250, and 9600], but none worked.

gtrainavicius, 2022-02-08:
USB-B jacks are not on the MIDI host side - you should be hooking up to USB-A host ports. Otherwise, the Arduino would have to be the one with a USB-A host port, it's much more involved compared to just acting as a USB peripheral device, so this mode of operation is unsupported by this library.

Also, UNO does not natively support non-COM serial communication through its built-in USB port, the workaround for such devices is to emulate USB in software through the GPIO pins and a small electronics circuit. This usage requires some knowledge of how to configure the library and build the circuit, so it's more straightforward to just use Arduino boards based ATMEGA32U4 controllers like SparkFun ProMicro, or something that already has the USB circuit for an ATTiny controller like DigiSpark Micronucleus.

HanalogInstruments, 2022-03-08:
Try this method to working as v-usb

[image]