BlokasLabs/USBMIDI #22: PluggableUSB.h error on SAMD

marchingband, 2022-08-17:
I am getting an error that I cannot resolve.

```
Arduino/libraries/USBMIDI/src/usbmidi_pluggableusb.cpp:13:10: fatal error: PluggableUSB.h: No such file or directory
 #include <PluggableUSB.h>
```
on line 13 your code is
```
#include <PluggableUSB.h>
```
other libs like MIDIUSB do something like
```
#if defined(ARDUINO_ARCH_AVR)
#include "PluggableUSB.h"
#elif defined(ARDUINO_ARCH_SAM)
#include "USB/PluggableUSB.h"
```
I am using Seeed XIAO SAMD, but I get the same compiler error using any SAMD board, Arduino Zero, Adafruit, or Seeed Studio.
I have v1.1.6 of USBMIDI, and version 1.8.19 of Arduino IDE.

gtrainavicius, 2022-08-24:
Could you edit the library to include the PluggableUSB.h as you showed?

Also check out the `sam` branch - https://github.com/BlokasLabs/USBMIDI/tree/sam

I haven't personally run the code on a SAM based Arduino, but if the PluggableUSB are compatible with AVR one, it might just work.

marchingband, 2022-08-24:
Ahhhh yes my bad, 3 years later I tried to reproduce the build, and forgot there was a SAM branch. Thank you!

gtrainavicius, 2022-08-24:
Does the SAM branch work fine for you?

marchingband, 2022-08-24:
I'm away on vacay now, but I'll try it in a few weeks. It def worked a few years ago.