BlokasLabs/USBMIDI #23: Defining custom vendor/product name fails

jsiddall, 2023-10-26:
I tried defining a custom vendor name using the example in usbmidi.h:

#define USBMIDI_DEFINE_PRODUCT_NAME('b', 'l', 'o', 'k', 'a', 's', '.', 'i', 'o')

but get:

`Arduino/libraries/USBMIDI/src/usbmidi.h:77:37: error: "'b'" may not appear in macro parameter list
 #define USBMIDI_DEFINE_PRODUCT_NAME('b', 'l', 'o', 'k', 'a', 's', '.', 'i', 'o');`

I also see a note "This works only on V-USB based implementation for now." but I am trying this on a Leonaredo-like board.

Any ideas about how I can define my own vendor and product names?

gtrainavicius, 2023-10-27:
Hi, `#define` is only necessary when you're defining the macro. When using it, you should do it without the keyword. For example, see here:

https://github.com/BlokasLabs/Midimon/blob/72dca09dc65414355744b8c827a70969a7c7db35/midimon.cpp#L32C64-L32C64

Use a line as such:

```
USBMIDI_DEFINE_PRODUCT_NAME('b', 'l', 'o', 'k', 'a', 's', '.', 'i', 'o');
```

jsiddall, 2023-10-27:
Gaah, thanks for setting me straight.  I had tried using the macro (without re-defining it) in it's own .cpp file as a test but, looking back, I failed to include usbmidi.h in that file.

Now, having got it successfully compiled and uploaded, I still just see the default device info:

```
Oct 27 09:30:57 kernel: [855326.412257] usb 1-7.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Oct 27 09:30:57 kernel: [855326.412260] usb 1-7.3: Product: Arduino Leonardo
Oct 27 09:30:57 kernel: [855326.412262] usb 1-7.3: Manufacturer: Arduino LLC
```

Interestingly, the MIDI device name seems to be something different, with "MIDI 1" added:

```
$ amidi -l
Dir Device    Name
IO  hw:4,0,0  Arduino Leonardo MIDI 1
```

Given this hardware, is there any way to change any of those values?

gtrainavicius, 2023-10-27:
It's just something that Linux's USB MIDI driver does on its own accord. :)

jsiddall, 2023-10-27:
Yeah, I figured ALSA was probably just enumerating it based on the product and device type.

Is there any hope for changing the product or manufacturer on an ATmega32u4?  Is that buried in a bootloader or something?

gtrainavicius, 2023-10-27:
If it's using the Pluggable USB implementation of USB MIDI, then it can't be changed from within USBMIDI.

If it is Pluggable USB, then it's defined somewhere in the board support files.

jsiddall, 2023-10-27:
According to the "other" MIDI USB project:

> Furthermore, since PluggableUSB only targets boards with native USB capabilities, this library only supports these boards (eg. **Leonardo**, Micro, Due, Zero and so on)
> For more information about this library please visit us at http://www.arduino.cc/en/Reference/MIDIUSB

So, sounds like it should work.  I will see if I can dig some more.

jsiddall, 2023-10-27:
I found USBCore.cpp has:

```
#if USB_VID == 0x2341
#  if defined(USB_MANUFACTURER)
#    undef USB_MANUFACTURER
#  endif
#  define USB_MANUFACTURER "Arduino LLC"
```

...but changing that value and recompiling had no effect.

boards.txt also has a line in the Leonardo section

`leonardo.build.usb_product`

but again, changing had no effect.  Hmmm...

Other bootloaders (ex: gemma) do make references to changing USB information (see: `usbconfig.patch`) but I am not sure if that would even work.  Pretty sure I don't want to dive that deep.

gtrainavicius, 2023-10-30:
After changing boards.txt I think you probably have to restart the Arduino IDE for changes to take place, at least I think I had such experience myself a little while ago when developing the boards.txt for Midiboy. This is outside of scope though of this issue on USBMIDI :) I think the best place to look for further assistance is the community forums of the particular Arduino board you're using.

jsiddall, 2023-10-31:
> After changing boards.txt I think you probably have to restart the Arduino IDE for changes to take place, at least I think I had such experience myself a little while ago when developing the [...]

Yes, you are correct, restarting and re-uploading worked for changing the usb_product in boards.txt.  Thanks for the tip!